commaai/opendbc #1848: Volkswagen D harness

andiradulescu, 2025-02-25:
|Terminal|MQBevo|Harness Mapping
| ------------- | ------------- |------------- |
|5|Switched 12V|Switched 12V
|8|Ground|Ground
|9|Fusion Ethernet Low|Passthrough
|10|Fusion Ethernet High|Passthrough
|12|ADAS CAN High|CAN 0/2 High
|13|ADAS CAN Low|CAN 0/2 Low
|14|Camera heater supply|Passthrough
|16|Camera heater ground|Passthrough

Connector seems to be 1K0_972_928_A
https://www.skoda-parts.com/spare-part/1k0972928a-flat-contact-housing-16-pin-skoda-45118.html
https://www.aliexpress.com/item/4001360821152.html 

[Camera Skoda Kodiaq 2.pdf](https://github.com/user-attachments/files/18964224/Camera.Skoda.Kodiaq.2.pdf)

[image]
[image]
[image]

[image]

1okko, 2025-03-24:
[image]

1okko, 2025-03-24:
https://item.taobao.com/item.htm?abbucket=11&id=850676560202&ns=1&pisk=gkt7MixDj7VWjUILdaDVhxUH_uICKxoaNJ6pIpEzpgIR9E9MO_knrg8fOIdfwukorwCBZgQHz65POM9H1xlZbcJkEMjx_foN_i_Z3GqL2uFFH-B1ATnS0qkJEMjKO-rK7c9k6WLxsYXdHx6FLkeJ9TFYHsXUy6ddwZFACOjdvbH-iwKMSQ-3v4QVOdBOa97H4wQCBe5hpuvhvxAJ0u7eTM0duF1lEJX-w2CAi9XLe6QKDoBOBWILvTpYkOBayyIdJKQAi9Id9BIKhZCcpjIxGt05wLMRfD0ULxiXHsZLvn6xt_p7xkXvATs51LpUquDhFa19eshZIiDCkpbBxJqCzLLMtOd8wbsphOO1CQgUK_TvQ38pFmGO43vJWwKttyd1VB_92ZwLALjPOi6vXvzGgn1y93_xQl_FDh7O2EumawS5CBKkNJZp6KJwqZxI1SCHzOjAFBntGg5bb1NEeywfSk6f_xMb-y03yfZBi5yoVaBc3hkjh7ahytXf_xMb-ybRntRthxN7-&priceTId=213e04ec17427757335762683e10c8&skuId=5884930170097&spm=a21n57.1.hoverItem.2&utparam=%7B%22aplus_abtest%22%3A%227241249c3a5ac152163af743e761455b%22%7D&xxc=taobaoSearch