commaai/opendbc #1899: Hyundai: split CAN and CAN FD CarController update function

sshane, 2025-03-03:
The entirety is very confusing, especially all of this: https://github.com/commaai/opendbc/blob/bc839875bd0ca476be71dea2cbebc92a4544c40c/opendbc/car/hyundai/carcontroller.py#L105-L164

and this function: https://github.com/commaai/opendbc/blob/bc839875bd0ca476be71dea2cbebc92a4544c40c/opendbc/car/hyundai/carcontroller.py#L174

`use_clu11` has nothing to do with CAN FD, but it looks like it could

sshane, 2025-03-03:
Related: https://github.com/commaai/opendbc/pull/1898, we weren't testing one of the ACC cancel messages, allowing actuation

sunnyhaibin, 2025-03-07:
`use_clu11` is required in CAN FD for these 2 ports:
- https://github.com/commaai/opendbc/pull/1268
- https://github.com/commaai/opendbc/pull/1273

It was introduced in this PR as prerequisites for the above 2 PRs:
- https://github.com/commaai/openpilot/pull/29663

sshane, 2025-03-12:
I see, you can make them separate functions `create_can_button_messages` instead of an argument to make it more clear

closed 2025-03-12 by sshane