commaai/opendbc #1992: Canadian 2022 Hyundai Tucson waiting for controls to start…

PatrickRodriguz2, 2025-03-13:
Hi all, I transferred my Comma 3 from my 2022 Elantra to my 2022 Tucson Hybrid. Both are Canadian cars. Open pilot worked perfectly in my Elantra however I can’t get it to work on my fully supported Tucson hybrid. I bought a new harness kit from open pilot and I still get stuck on “open pilot unavailable, waiting for controls to start.” I tried different forks, same issue. Clean install and resets. No help. This is the second harness I’ve used. Only thing that hasn’t changed is the USB C to USB C cable. I’m genuinely stumped.

sshane, 2025-03-14:
What is your dongle id?

PatrickRodriguz2, 2025-03-14:
I'll get back to you but someone commented saying I need a redpanda for it because it's a Comma 3 and not a 3X.

sshane, 2025-03-24:
If you use a 3X/red panda and still experience issues you can report back.

closed 2025-03-24 by sshane