commaai/opendbc #1994: Controls mismatch alert and subsequent disengage

MrXelious, 2025-03-14:
This issue was originally posted under OpenPilot #34677 and moved here per request.
---

Seemingly random controls mismatch alert and subsequent disengage.

I have been unable to notice a rhythm this behavior. Sometimes it happens very quick after engagement, while other times it takes some time.

Sometimes it happens once then calms down while other times it will happen repeatedly in quick succession. 

Operation had been rock solid on older builds. I have not pinned down at which version the issue started.

The system can immediately be re-engaged after alarm.
2015 Hyundai Genesis
3687c15bdfe09295/0000005f--888dec6063
Release 0.9.7

sunnyhaibin, 2025-03-15:
Safety is sporatically disallowing controls, triggering controls mismatch:

[image]

sunnyhaibin, 2025-03-15:
Safety is sporadically disallowing controls due to invalid rx checks, triggering controls mismatch.

@sshane we may have more rx checks missing, specifically for Hyundai legacy cars 😅

[image]

sunnyhaibin, 2025-03-15:
Safety replay is showing invalid rx for `1057|0x421|SCC12`:
```
replaying 3687c15bdfe09295/0000005f--888dec6063 with safety mode 23, param 0, alternative experience 1

RX
total rx msgs: 6226045
invalid rx msgs: 8
safety tick rx invalid: True
invalid addrs: {1057}

TX
total openpilot msgs: 160558
total msgs with controls allowed: 117230
blocked msgs: 1442
blocked with controls allowed: 0
blocked addrs: Counter({832: 1433, 1265: 9})
```

sshane, 2025-03-16:
Looks like this isn't a regression, @MrXelious have you ever hit this before?

MrXelious, 2025-03-16:
> Looks like this isn't a regression (happens on 0.9.7), @MrXelious have you ever hit this before?

Negative. Operation had been rock solid for quite some time. I ended up a few versions behind and this issue popped up after getting back up to speed.

sshane, 2025-03-16:
Looks like our Python and safety calculation match, the module just sent an incorrect checksum for SCC12:

```
> msg = bytearray(b'\x00 \x00\xff\xe3\x7f!\xf8')
> actual = msg[7] >> 4 # checksum
> msg[7] = msg[7] & 0xf  # remove checksum
> python_expected = 0x10 - sum(sum(divmod(i, 16)) for i in msg) % 0x10
> safety_expected = hyundai_compute_checksum(1057, msg)
print('python sums', [sum(divmod(i, 16)) for i in msg])
print('python_expected', python_expected, 'safety_expected', safety_expected, 'actual', actual)
```

sshane, 2025-03-16:
We don't have signal defintions for these, but two AEB related signal (one called `AEB_Failinfo`) that activated shortly after SCC12 sent two duplicate checksums with decrementing counters and all other signals the same:

[image]

I would bring your car into service, the ECU looks to be defective.

closed 2025-03-16 by sshane