commaai/opendbc #2065: BYD Atto 3 Harness

qzwf, 2025-03-29:
- [ ]     which car is this for? 2022-2024 BYD ATTO3
- [ ]     schematic - Push to neo repo once confirmed working
- [ ]     beta sent to tester
- [ ]     in shop
- [ ]     Added to openpilot docs

The ADAS Camera in BYD ATTO3 is MVS4 from Veoneer. Mono Vision System 4 . Will assign proper harness name once more information is obtained.

qzwf, 2025-03-29:
[image]

Harness Wiring as per Service Document of BYD Atto 3 

[image]

[image]

Connector Images

[image]

[image]

The connector used in BYD is TE 1-1355211-2 
https://www.te.com/en/product-1-1355211-2.html
Mating Part
https://www.te.com/en/product-1-1670459-2.html

qzwf, 2025-03-29:
| **BYD Connector Pinout**  | **Comma 3x Connector Pinout (18pin ) V3** |
|---------------------------|-------------------------------------------|
| 1 - GND                                | 1,17                                      |
| 2 - IGN1                                | 2,14                                      |
| 3 - Private CAN_H               | 4,16                                      |
| 4 - Chassis Network CAN_H | 8,9                                       |
| 7 - Private CAN_L                  | 6,18                                      |
| 8 - Chassis Network CAN_L | 7,11                                      |

Based on the above, we can determine the pin layouts as above.

qzwf, 2025-03-29:
[image]

adeebshihadeh, 2025-03-29:
Looks good! A few questions:
* Does this support openpilot longitudinal control?
* How many BYD models does this support?