commaai/opendbc #2077: Unsupported Car Notification - 2022 Hyundai Kona

leeadamp, 2025-04-04:
Hi, I recently got a 2022 Hyundai Kona Limited with smart cruise control, so seems to be on the supported list. I'm having some issues getting it working. I posted over the in installation help, and Marco tried to help me out, but pointed me to Hyundai channel and haven't received a specific response there. My name on discord is Bartgoo and you can see my questions in both channels.
Here is a short summary of my issue:
I am getting the unsupported car notification. I used the harness and usb-c cable that came with the C3x I just purchased directly from comma. I made public routes from the last couple weeks if that helps anyone.
a0084b9a0949d2c0/00000001--ed1fdafb54
a0084b9a0949d2c0/00000002--abe5d31834
While I'm not an expert by any means, I wonder if this car has the correct fingerprinting? Looking at the values for in opendbc hyundai kona 2022 fingerprints, it seems like the fwdCamera, eps, and fwdRadar values are aligned, but the transmission values aren't. Marco helped me create a new fork with just the updated transmission value, but that didn't seem to work.
Please let me know the best way to move forward with this.

Best,
Adam

sshane, 2025-04-08:
Try this installer URL:

https://installer.comma.ai/commaai/kona-2022-fp

sshane, 2025-05-07:
Can you drive more on the highway so we can verify the lateral tuning?

leeadamp, 2025-05-07:
Will do, I have at least one in the system and I'll upload tomorrow

leeadamp, 2025-05-10:
@sshane hi, I've made public 6 different routes using the updated installer you made. Let me know if I can provide any help.

closed 2025-06-04 by sshane

leeadamp, 2025-08-22:
@sshane Any chance you are able to reopen the previous fork you made? I just installed the current version of openpilot and it doesn't seem to support my car anymore.
Thanks so much.

jyoung8607, 2025-08-22:
I checked the device you were using when you opened this PR. All the routes uploaded today on stock openpilot recognized your car. Can you provide a specific route and tell us specifically what you're seeing?

leeadamp, 2025-08-22:
a0084b9a0949d2c0/00000003--71d9d39315

Thanks for the reply. The above route is likely one you looked at already. It recognized the car, but wouldn't move out of dashcam mode. Looking at the support info for knows cars it seems that Kona 22-23 now only shows support in dashcam mode. I wonder if there was some sort of broader change that removed support for it.