commaai/opendbc #2084: VW: Software triggered ABS/Cruise Control fault

airsky98, 2025-04-08:
### Describe the bug

I'm driving a VW tiguan 2020 in Asia. 

[ad5efcc390e8d37e/0000001c--0d94bb6d2d/0](https://connect.comma.ai/ad5efcc390e8d37e/0000001c--0d94bb6d2d)

It seems like the software will trigger an ABS/cruise control fault while the open pilot is active. If I restart the car without turning on the open pilot to take over the steering wheel, such issue won't occur for the rest of the journey. 

### Provide a route where the issue occurs

ad5efcc390e8d37e/0000001c--0d94bb6d2d/0

### openpilot version

agnos11.12 

### Additional info

i'm on this version in order to fix the frequent restart of the unit

jyoung8607, 2025-04-08:
None. I queued rlogs for upload earlier today but the device hasn't been on Wi-Fi.

jyoung8607, 2025-04-08:
Found that partial rlogs were already uploaded for one of the two routes, and they cover the reported event.

@airsky98 Your Tiguan's front-right ABS wheel speed sensor is bad. It's producing a noisy signal with intermittent dropouts, and eventually your ABS/ESP stops being able to get a signal at all. When this happens, it will produce a variety of other faults in the car, including with ACC. This is not a problem with openpilot; it would happen even without openpilot installed. You'll need to take it into a dealer for repair.

[image]

[image]

jyoung8607, 2025-04-08:
I pondered leaving this open, re-titled to add Q-bit support for various signals (like the wheel speed sensors) like Ford does. However, the TSK will (and did) fault for this sort of thing, which means we already have effective coverage of sensor quality faults with `accFaulted`.

[image]

closed 2025-04-08 by jyoung8607

airsky98, 2025-04-09:
thanks! will replace the sensor and try again .

airsky98, 2025-04-10:
is the tool used above open to public or it is an internal only tool?

jyoung8607, 2025-04-10:
> is the tool used above open to public or it is an internal only tool?

If you're just curious how I arrived at this conclusion, the tool is [Cabana](https://github.com/commaai/openpilot/blob/master/tools/cabana/README.md), which is open source. It looks at CAN message logs from an uploaded openpilot route log. You can use it after setting up an [openpilot dev environment](https://github.com/commaai/openpilot/blob/master/tools/README.md) on your machine.

Your first route only has partial rlogs uploaded, so you'd launch Cabana with the route `ad5efcc390e8d37e/0000001c--0d94bb6d2d/41:64` to look at just segments 41 through 64. The fault happens about 60% along the timeline. You can look at `ESP_19` for the individual wheel speed signals, and `ESP_10` for the underlying sensor impulse count integrals, directions, and fault indicators.

When actually trying to repair the car, the correct tool is a OBD/UDS diagnostic tool like VCDS or OBDeleven. There will be faults stored in the ABS module at address 03. Or, just drop it at your preferred dealer or auto repair shop if you don't want to work on it yourself.

airsky98, 2025-04-21:
Got it. It works well after replacing the front right abs sensor. 

Just curious on these live messages as well. Will setup one when I have time. Thanks again ! 👍