commaai/opendbc #2146: 2025 Elantra hev car Unrecognized

prokary91, 2025-04-23:
Describe the bug
2025 Elantra hev car Unrecognized

Which car does this affect?
2025 HYUNDAI Elantra hev

Provide a route where the issue occurs
[d01e5725e5587ae3/00000001--a4126f0d93/0
](https://useradmin.comma.ai/logs?url=https%3A%2F%2Fcommadata2.blob.core.windows.net%2Fqlog%2F2f61bea2cb9328f6%2F00000004--10b6de397f%2F0%2Fqlog.zst%3Fse%3D2025-04-23T08%253A52%253A48Z%26sp%3Dr%26sv%3D2024-08-04%26sr%3Db%26rscd%3Dattachment%253B%2520filename%253D2f61bea2cb9328f6_00000004--10b6de397f--0--qlog.zst%26sig%3DVPgLjjjEZo%2FWuapxeMV%2FRwxg6pMEd4WPBnxSko6%252B9sg%253D&event=carParams)

openpilot version
0.9.8

Additional info
It can be confirmed that the vehicle is a CAN and a camera SCC

royjr, 2025-04-26:
We have a PR for this https://github.com/commaai/opendbc/pull/1537. You can install it with `royjr/elantra-2024-port`. You can report any issues in this thread https://discord.com/channels/469524606043160576/1311332559816491009
`

prokary91, 2025-04-26:
Thank you for your response. However, I’m not sure how to actually apply this. Could you provide a download link for something that can be applied to the comma3X? Also, I entered the Discord link, but it says I don't have permission to send messages in that channel

> We have a PR for this [#1537](https://github.com/commaai/opendbc/pull/1537). You can install it with `royjr/elantra-2024-port`. You can report any issues in this thread https://discord.com/channels/469524606043160576/1311332559816491009 `

royjr, 2025-04-26:
Try joining the discord first. https://discord.comma.ai

prokary91, 2025-04-26:
I have installed your branch and confirmed that it recognizes my car. 
Thank you for your help. 
I will check if it works tomorrow