commaai/opendbc #2156: OP does not engage aftre pressing "Cruise" and then "Set"    takes multiple presses of the "Set" button to engage.

stevegf3675, 2025-04-24:
Hyundai Genesis 2016 AWD / No SMDPS

Comma3

0.9.8-release / releasec3 / 5a84ebb / April 16

installed 2025/04/24 on my Comma3

After pressing the "Cruise" button and then pressing "set," the SCC does not engage and requires multiple presses to engage.

Preserved routes with this behavior :

27a580d124055206/00000000--b81083aa2e
27a580d124055206/00000001--3399a5981f
27a580d124055206/00000002--3748f9144e

sshane, 2025-05-07:
Does it say anything when you press SET?

stevegf3675, 2025-05-07:
There are no messages. The only icon visible on the dashboard is the SCC icon, which flashes each time you press the “set button.” Most of the time, it requires a quick double-press to engage OP.

sshane, 2025-05-13:
This will be fixed in https://github.com/commaai/openpilot/pull/35207

closed 2025-05-18 by sshane