commaai/opendbc #2158: BUICK lacrosse 2021 openpilot UnavailableRadar Error: Restart the Car

zhou8603051, 2025-04-25:
### Describe the bug

BUICK lacrosse 2021 openpilot UnavailableRadar Error: Restart the Car    OH NO

### OS Version

ubuntu 24.04

### Additional info

BUICK lacrosse 2021 openpilot UnavailableRadar Error: Restart the Car    OH NO think you

sshane, 2025-05-07:
Please provide your dongle ID. Does your car have a radar and adaptive cruise control?

zhou8603051, 2025-05-07:
pc openpilot

发自我的iPhone

------------------ Original ------------------
From: Shane Smiskol ***@***.***&gt;
Date: Wed,May 7,2025 2:15 PM
To: commaai/opendbc ***@***.***&gt;
Cc: zhou8603051 ***@***.***&gt;, Author ***@***.***&gt;
Subject: Re: [commaai/opendbc] BUICK lacrosse 2021 openpilot UnavailableRadarError: Restart the Car (Issue #2158)

sshane, 2025-05-07:
Does the car itself have adaptive cruise control?

loggerd writes out logs to your PC, you can send me those. Usually at ~/.comma/media/0/realdata

zhou8603051, 2025-05-07:
ok think you  

发自我的iPhone

zhou8603051, 2025-05-07:
It has ACC adaptive cruise control.

发自我的iPhone

sshane, 2025-05-07:
Without logs, we cannot do much else to help. A photo of the setup might also be useful. Some other issues could be using an improper USB C cable with the harness (should be 3.1 gen 2 or similar)

closed 2025-05-07 by sshane