commaai/opendbc #2160: 2018 Hyundai Sonata gets frequent AEB/LKAS errors since 0.9.8

Jon8RFC, 2025-04-25:
My 2018 Hyundai Sonata (SONATA_LF) gets frequent AEB/LKAS errors since updating to 0.9.8.

https://youtube.com/shorts/e_p5m_zWgZk

Example routes:
c1fc13c1b806e536/00000219--674500b4d2
c1fc13c1b806e536/0000021a--0b2b57f4fb
c1fc13c1b806e536/0000021b--50525c5018

Unsure if it's related, but when those errors begin I think I hear one of the comma3 cameras getting noisy.

sshane, 2025-05-07:
Does this only happen on startup, and once it goes onroad (shows the camera), it goes away?

Jon8RFC, 2025-05-07:
No.  It will happen seemingly randomly after startup
It always happens on startup, but the frequency of repeated errors after startup I can't figure out a pattern.
Restarting the car will have the errors pop up even with the Comma3 is already booted.
Power cycling with 30+ seconds before restoring power on the Comma3, with the car still running, results in the errors once the Comma3 boots up.

It can also go 10 minutes without any errors during a drive, then an error will pop up on the car
Sometimes it will go an entire drive without errors after it calms down for 10 minutes of errors.

⠀

If it's helpful (since @adeebshihadeh said no errors show in the logs for the Toyotas), I can buy a mount for my phone to setup in my car, then if you make a dev version which has an on-screen timestamp (HH:mm:ss:mmm), then I can record a drive with my phone having the comma3 & dash in view. You can see when the errors on the car happen, and use the Comma on-screen timestamp to refer to the logs coinciding with the timestamp. If the Comma3 always has an HH:mm:ss:mmm display going, which correlates to the logs on the device, then you can just look at the video recording I made and match up the timestamp to the logs.  Or whatever else you think might be helpful. That could stick around as a setting toggle for future edge-case debugging scenarios like this one.

jyoung8607, 2025-05-07:
Something is wrong with CAN2. We seem to have trouble with TX, lots of busOff events. We have different trouble with RX, we don't seem to be outright losing packets from the camera but there's some timing issue, some 100Hz LKAS_11 messages are doubling up and then skipping 100Hz `card` frames.

Effects of this seem to spill over to CAN0, openpilot struggles at times to get LKAS11 on the wire at 100Hz. This will definitely produce a LKAS Fault. The behavior is probably described by commaai/panda#968 but triggered by the CAN2 problem.

[image]

A physical fault with the bus could explain this. It's possible we're getting all traffic *from* the camera successfully since we're the only other device on the bus, so the camera retransmits until Panda ACKs it, but we're dropping traffic *to* the camera since we're forwarding a great deal more traffic and we don't handle queue backups all that well, and as mentioned above that effect can spill over to other buses.

I'd try changing out your OBD-C cable first, and if that has no effect on the problem, let's see some deliberate A/B testing with release 0.9.7 vs release 0.9.8 to see if there's truly a software connection.

[image]

[image]

Jon8RFC, 2025-05-08:
@jyoung8607 Thanks for the insight and analysis--I didn't realize those were from my routes and assumed they were examples from another.

I wasn't suspecting a cable since it still could self-drive and cleaned up after a while, haha.  I don't have a compatible cable to troubleshoot with, but testing on 0.9.7 showed the same two AEB and LKAS errors, plus an automatic lights error.  Just a perfect storm.  My battery died when 0.9.8 came out and I assumed it was some BCM fluke due to the persistent drain messing with the car (I had a BCM issue in the past from a third-party OpenPilot)...after a while, I blamed 0.9.8 but it's probably not 0.9.8 afterall.

It all goes away when I unplug the Comma3, so you're probably right.  I unplugged the harness for a couple of minutes as well since I don't know if there's anything in there other than wiring or a PCB...issue persisted.

I ordered the [Discord-suggested cable from Amazon](https://www.amazon.com/dp/B07VMKRKBR) and if it's a no-go, I'll order a Comma cable.  It'll arrive Friday and then I can close or keep this issue open.

Jon8RFC, 2025-05-11:
Same errors with the amazon cable. I hope it's not a comma/harness issue, and hope maybe it's firmware installed with 0.9.8.

I ordered some contact cleaner and I'll order a comma cable to rule it out.

Jon8RFC, 2025-05-17:
I got slightly different behavior after trying the old comma cord and spraying contact cleaner. When I started it up without the obd power, I had zero errors, and the next drive was also zero errors. I turned it back on and will drive again after it's done a shutdown and also connect the new comma3 cable when it arrives.

The clicking turned into a single click as well, even when it was throwing errors.

It may have been contact cleaner and the connecting/reconnecting of cables cleaning up the pins. We'll see.

Jon8RFC, 2025-05-21:
I sprayed more contact cleaner and inserted/removed the USB cable multiple times in both orientations.

Still a single click, no more multi-clicks, zero errors, and it started from a cold boot at car start today and I'm using the same comma3 cable that's 3 years old. 

Contact cleaner and multiple insertions/removals did the trick. Sorry for the false alarm.

closed 2025-05-21 by Jon8RFC