commaai/opendbc #2166: Toyota intermittent faults on startup

adeebshihadeh, 2025-04-29:
From this [Reddit thread](https://www.reddit.com/r/Comma_ai/comments/1k9kwr1/comment/mph50wx/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button). There's no faults in the logs, so it seems there's some fault we don't currently catch/parse out.

```
a447729c1d15ff89
```

jacklee206, 2025-04-30:
Happens a lot after c3 cold start , relay randomly clicks in before onroad service started and car throws error after losing connection to the ADAS camera. Here's a viceo i took when this happend [https://discord.com/channels/469524606043160576/871838269405556736/1363983234781675560](url)
possible clue: problem goes away by connecting to an external black/red panda

codyisapro, 2025-04-30:
dd39308405abab68

Similar issue, sometimes clears on start up of openpilot, sometimes requires restart of vehicle multiple times.

8a9phrywet978ui, 2025-05-02:
ad189098ce1794e3

Copy/pasting from Discord:

within the last month or two started happening with my '19 Corolla Hatchback with Sunnypilot, both dev-c3-new and staging-c3-new. A mix of it throwing LKAS faults on boot before it self-clears, and sometimes throwing every fault imaginable while driving, requiring an ODB Clear and Car restart to fix

Verified everything is properly seated and reseated 

Per Erich:

Confirmed it happens with and without (Comma Power Connected) for the on-boot lkas fault (comma completely off, or on with car off doesn't seem to matter). The full dash faults this morning, comma power was connected, need to verify it also happens with it disconnected but pretty sure it does (referring to an instance probably 3? weeks ago)

dev logs screen is blank in sp, not sure if that's relevant

Jon8RFC, 2025-05-06:
> Happens a lot after c3 boots from a cold start , relay randomly clicks in before onroad service started and car throws error after losing connection to the ADAS camera. Here's a viceo i took when this happend [https://discord.com/channels/469524606043160576/871838269405556736/1363983234781675560](url) possible clue: problem goes away by connecting to an external black/red panda
> 
> afa81bac0f6fc943

Same for me on my Hyundai Sonata.  I assumed it was the camera, but that sounds the same. as mine.  My issues began in 0.9.8.  I didn't take a video of the boot up, but I should.  Thanks for sharing that.
https://github.com/commaai/opendbc/issues/2160

adeebshihadeh, 2025-05-06:
Looks like everyone here has a comma three...

devimauz, 2025-05-07:
+1 with comma three and same error.

sshane, 2025-05-07:
Is this related to flashing panda on startup?

sshane, 2025-05-07:
Can someone share a specific route where this happens on Toyota so I can check if a fault signal I know of captures this?

jacklee206, 2025-05-07:
> Can someone share a specific route where this happens on Toyota so I can check if a fault signal I know of captures this?

afa81bac0f6fc943/00000048--027e2f2823
afa81bac0f6fc943/00000049--dfb749ee21

8a9phrywet978ui, 2025-05-07:
Adding to previous comment, did a full reinstall of everything after installing an SSD, and on sunnypilot's staging-c3-new branch, it's now throwing Cruise Fault every single time, regardless of ODB power circumstances. Previously it was intermittent, and that was a build from a week or two ago at most.

Reinstalled stock OP and it works fine.

ad189098ce1794e3
2019 Corolla Hatchback

codyisapro, 2025-05-09:
> Can someone share a specific route where this happens on Toyota so I can check if a fault signal I know of captures this?

dd39308405abab68/0000069f--f8e5cd23b8/0

Typically happens after a short time powered off, like stopping for gas. (No comma power)

nelsonjchen, 2025-05-10:
My Corolla was complaining about all sorts of LKAS, RSA, and etc errors like the camera was disconnected when starting up. This has happened two or three times. This is sort of in the pre-onroad mode from cold boot. 

I also noticed that when the car started up, the relays made an awful like rapid clicking sound. It's like a short angry cicada noise. I'm not sure if that's software or a hardware issue. The errors kept on coming in the dash until the device started up into onroad.

I do have a comma power installed.

One of the routes I have is here where the issue manifested before:

https://connect.comma.ai/fe18f736cb0d7813/0000006b--5849e229e5/1/250

[image]

devimauz, 2025-05-11:
> [Követő kép][https://github.com/notifications/beacon/ACUPEPQWFGALKLVA4BT23WD25Z6SRA5CNFSM6AAAAAB4DUWPW6WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTVLAREQU.gif]

Its not hardware issue. Mine doing the same. Relay clicks and propably that makes the fault. OBD Power plugged in. You wont see anything in routes because.

choongsoo, 2025-05-11:
Comma 3 just finished booting and the dashboard showed all sorts of faults (LTA, RCTA, Front Camera, SRS Airbag) pre-onroad. Once it switched to onroad, all the cruise/camera related faults went away but the SRS airbag fault persisted. Restarting the car reset the airbag fault.

The route recorded right after that boot is a447729c1d15ff89/00000074--d882c004b5.

Like nelsonjchen, I also hear some rapid clicking noise sometimes, but I am not sure if I heard it on this particular occasion.

Thanks for investigating this issue!