commaai/opendbc #2228: GM: Car enters ACC standstill while in motion

nworb-cire, 2025-05-13:
If the car is at ACC standstill and the resume button is pressed before OP wants to send a gas command, the car will (expectedly) lapse back into standstill after a moment. It is possible to time this just wrong such that the brakes release but the car falls back into ACC standstill. I believe edge cases of gas gating may also be able to cause this, though I have not observed it happen. No fault is displayed, but the car is unresponsive and in motion until the driver resumes input.

```
806907cd33ef2647/00000070--6350d749b6/5
```

sshane, 2025-05-13:
I think the EBCM accepts braking commands in this state, we should probably brake to a stop when in standstill. Bug in longcontrol?

I've had it happen where if we don't request enough gas for a steep hill such that the car just stays in place, the ECM eventually enters standstill mode, and we continue to try to apply gas which is no longer actuated and we roll back indefinitely.

sshane, 2025-06-17:
Need a route to look into complaints.