commaai/opendbc #2268: Move MISRA static analysis into `./test.sh`

adeebshihadeh, 2025-05-18:
`./test.sh` is supposed to be the all-in-one script for this project, but some tests are currently split out since they're too slow for our 1m CI timeout limit. `opendbc/safety/tests/misra/test_misra.sh` is one of those.

This is an exploratory bounty. Maybe it's one stupid thing that's slow and a CLI arg will fix it. Or maybe it's a bunch of small things to fix in cppcheck. To claim this bounty, root cause why this is so slow and come up with a plan to make it fast enough to put in `./test.sh` without hitting the CI timeout. Once we know how much work it is, we can assign a follow-up bounty for actually shipping the speedup.

NoahCristino, 2025-05-19:
Why not run cppcheck with -j <jobs> to multithread it (https://linux.die.net/man/1/cppcheck)

adeebshihadeh, 2025-05-19:
Maybe that's all we need! Try it out and come up with a plan if you'd like to lock the bounty.

bravochar, 2025-05-20:
I took an initial look, and it appears that both `--cppcheck-build-dir` and the `-j` option will only speed up your checks when you are checking _multiple files_, but the safety code is structured as single `.c` file that includes the rest of the safety code as `.h` files. As a result, I saw no speedup with those flags. Of the ~20 seconds that `test_mirsa.sh` takes to run on my machine, most of that is spent running `cppcheck` against `main.c`:

```
$ ./time_misra.sh
Resolved 67 packages in 6ms
Audited 56 packages in 0.53ms
HEAD is now at ac9db3069 2.17.1: Set version
make: 'cppcheck' is up to date.
scons: Reading SConscript files ...
scons: done reading SConscript files.
scons: Building targets ...
...
scons: done building targets.

** Safety **
Checking /home/charlie/opendbc/opendbc/safety/main.c ...
Checking /home/charlie/opendbc/opendbc/safety/main.c: __GNUC__=9...
nofile:0:0: information: Active checkers: 309/1056 [checkersReport]

real    0m7.851s
user    0m7.379s
sys     0m0.169s

** Safety with CANFD **
Checking /home/charlie/opendbc/opendbc/safety/main.c ...
Checking /home/charlie/opendbc/opendbc/safety/main.c: __GNUC__=9;CANFD=1...
nofile:0:0: information: Active checkers: 309/1056 [checkersReport]

real    0m8.560s
user    0m8.237s
sys     0m0.122s

Success! took 23 seconds
```

If the goal is for `.test.sh` to be a one-stop "Will my PR get a green check"-script, could the solution instead be for that script to also accept command-line arguments to do the individual tests with the default behavior of running them all? That way, the CI jobs could remain separate to stay under the 60 second limit while developers can run all the test from one place.

adeebshihadeh, 2025-05-20:
> If the goal is for `.test.sh` to be a one-stop "Will my PR get a green check"-script, could the solution instead be for that script 

So yes, that is the goal. However, I don't like the idea of splitting things up. It's really nice that CI literally just does `./test.sh` like you do on your PC.

I actually went ahead and did this with some hacks (moving to a faster CI machine and only testing the `CANFD` build https://github.com/commaai/opendbc/pull/2276), but it seems like you're already on the right track here. Interested in a bounty to move the MISRA mutation tests into test.sh?

varshith257, 2025-05-29:
I would like to take this issue forward snd complete

closed 2025-06-06 by adeebshihadeh