commaai/opendbc #2371: Comma 3X does not enable Lateral Control on Hyundai i30 (2018 Smart Trim with ACC)

Mustafa911991, 2025-06-05:
### Vehicle & System Info:
- **Car:** Hyundai i30 2018 (Smart Trim, factory Adaptive Cruise Control)
- **Comma device:** Comma 3X
- **OpenPilot fork tested:** `sunnypilot dev-c3`, `HKG_6.5`, and others
- **Harness:** Hyundai CAN harness (plugged into correct port)
- **Radar:** Built-in Bosch LRR

---

### Issue:
We are trying to get full Lateral Control support on this vehicle, including steering below 51 km/h. However, we encountered multiple problems:

- Comma 3X does not establish any steering control (Lateral Control never activates).
- Radar does not seem to initialize or communicate.
- SSH/ADB access does not work (public key was uploaded, device is silent).
- No debug logs can be retrieved due to no SSH access.

---

### What We Tried:
- Enabled Developer Mode
- Uploaded public/private keys
- Attempted to generate `.tar.gz` custom build
- Ensured CAN and power wiring is correct
- Tested multiple forks (Sunnypilot, official OP, etc.)

---

### Our Goals:
- Enable full Lateral Control, including steering under 51 km/h
- Get radar communication working properly
- Enable SSH or find a method to pull logs for debugging

---

### Questions:
1. Which OpenPilot branch/fork is fully compatible with this vehicle?
2. How can we override the 51 km/h steer limitation on Hyundai i30?
3. Why might Lateral Control be completely unavailable? Is MDPS connection required?
4. Can we retrieve logs or troubleshoot without SSH access?

---

Any help from developers or those with similar setups is greatly appreciated. Thanks!

sshane, 2025-06-06:
In your device settings there will be a dongle ID. Please post that. Connect your device to WiFi and it will upload logs

sshane, 2025-06-16:
I don't see any of the logs uploaded. I requested upload for https://connect.comma.ai/e46b08e501215981/0000001b--2f01562123 now. Please place your device on WiFi and ensure the rlogs upload. If the network is metered, you can toggle that option to unmetered in network settings

sshane, 2025-07-09:
The device hasn't been online for almost a month, so there are no logs we can look into still. Please comment back if you can upload the required logs

closed 2025-07-09 by sshane