commaai/opendbc #2383: Honda Fit, CAN Error: check connections

databaseddoctor, 2025-06-09:
Hi, please help me with this issue that has been preventing me from using my comma ai for the last few months.  I have already tried to reset the device and reinstall the software and the error still occurs.  Essential shortly after I start the device with my car, comma could give an error "CAN error: check connections:"  Every single light on my dashboard should come on and I won't be able to turn off the car without do a "hard push" on the start stop button.  My car is a 2019 Honda Fit and I cannot use my comma safely and it has been working for years!

[image]

[image]

[image]

vanillagorillaa, 2025-06-09:
If this install has previously worked, you may start with replacing the USB C cable to the device. Any off the shelf USB 3.1 Gen II or thunderbolt III cable will work

databaseddoctor, 2025-06-10:
What happened to the USB C cable?  I would really like to avoid replacing it...it took me many hours to install because all tight everything was.

vanillagorillaa, 2025-06-10:
Cables fail? 

You haven't provided a route with this issue for comma to review, but replacing the cable is a good first troubleshooting step

closed 2025-06-12 by adeebshihadeh