commaai/opendbc #2404: EV6: Can Error Take Control Immediately

nangelo1, 2025-06-17:
### Describe the bug

Generally, once during an hour-or-more-long drive on the highway, Openpilot will disengage, and I will get the message "Can Error, Take Control Immediately." Comma support advised me that this is a software issue. When it happens, I can immediately engage Openpilot again, and it doesn't typically happen more than once a drive. 

This is on a 2024 Kia EV6 running 0.9.8

### Provide a route where the issue occurs

2a4c6c4f64d7746d/00000015--6c7bff12dd/0

### openpilot version

0.9.8

nangelo1, 2025-06-18:
> [@nangelo1](https://github.com/nangelo1) can you switch openpilot to dashcam mode (openpilot toggled OFF in toggles), and drive around with your stock ADAS system for a few hours over a few routes? Want to double check that the stock system is tolerant to this

Sure, I'll try to do that in the next few days and report back.

nangelo1, 2025-06-30:
> [@nangelo1](https://github.com/nangelo1) can you switch openpilot to dashcam mode (openpilot toggled OFF in toggles), and drive around with your stock ADAS system for a few hours over a few routes? Want to double check that the stock system is tolerant to this

Not sure if this is enough data to capture the error, but here are a few routes with OpenPilot off. Happy to share more if the issue doesn't show up here. 

2a4c6c4f64d7746d/0000001c--2717a0e123/0
2a4c6c4f64d7746d/0000001b--dee38281d9/0
2a4c6c4f64d7746d/0000001a--76137ce3eb/0

jyoung8607, 2025-07-07:
> CRUISE_BUTTONS stopped sending for a second. No signs of issues from the panda from a quick look. No errors, no blocked msgs, no change in received msgs rate.

It looks like the car/sender's fault. When we start receiving again, COUNTER unexpectedly resets to zero. If those messages were dropped in the network, or somewhere internal to panda or openpilot, I'd expect the 50hz mod 0xF COUNTER to continue cycling every 300ms and therefore resume at 9 or 10. It looks like the actual sender reset.

For later reference, the gap is between time offset 568.049 and 568.460. The COUNTER timing pattern repeats on a 3-second cycle, so at-a-glance you can see what it should have been by looking at second 565 instead of 568.

adeebshihadeh, 2025-08-15:
Another case:
```
2fd850c60cc5bfef/00000121--0082c73da9
```