commaai/opendbc #2405: Honda Bosch A: 0.9.9 change is causing power steering failure

mvl-boston, 2025-06-18:
I've been debugging an issue with a 25 RDX loaner.  This issue causes the entire power steering system to shut down, which is very dangerous as the driver needs to steer with manual unassisted steering only, until they restart the engine.

It appears the following section in 0.9.8 honda safety was deleted in 0.9.9.

static bool honda_bosch_fwd_hook(int bus_num, int addr) {
  bool block_msg = false;

  if (bus_num == 2)  {
    bool is_lkas_msg = (addr == 0xE4) || (addr == 0xE5) || (addr == 0x33D) || (addr == 0x33DA) || (addr == 0x33DB);
    bool is_acc_msg = ((addr == 0x1C8) || (addr == 0x30C)) && honda_bosch_radarless && honda_bosch_long;
    block_msg = is_lkas_msg || is_acc_msg;
  }

  return block_msg;
}

It is causing relayed steer control messages to overlap with comma steer control messages.   And causes EPS faults.

Original 0.9.8 log without issues: ad9840558640c31d/00000005--1f77ad20f1
New 0.9.9 log with issues: ad9840558640c31d/00000086--362d0e6a98

Todo: making a backout fork for 0.9.9 to confirm this fixes the problem.

mvl-boston, 2025-06-18:
Closing - seems that this doesn't happen on devel 0.9.9.  Even with the same car support code.    Only (so far) seems to be a problem on master.

sshane, 2025-06-18:
The second route is on a custom fork quite a bit behind master, with opendbc changes. Is this car officially supported? What is a "backout fork?"

> Closing - seems that this doesn't happen on devel 0.9.9. Even with the same car support code. Only (so far) seems to be a problem on master.

> Why?

> It appears the following section in 0.9.8 honda safety was deleted in 0.9.9.

That's because blocking messages are now handled by a static tx msg config: https://github.com/commaai/opendbc/pull/2103

[image]

sshane, 2025-06-18:
All this said, I don't see any reason why this would happen. The change of now forwarding STEERING_CONTROL to the radar bus (bus 0) is new with that blocking config refactor, but it is a behavior difference with alpha longitudinal which is not in release yet. That said, I tested this on our Honda Accord (Bosch) with no issues, and it also makes sense since the radar is disabled and doesn't forward the camera's message to the PT bus. So this assertion is false:

> It is causing relayed steer control messages to overlap with comma steer control messages. And causes EPS faults. These usually happen when a driver overrides steering while OP is engaged.

I don't think this is new in 0.9.9

mvl-boston, 2025-06-18:
> You also added back `honda_bosch_fwd_hook`, I can see there easily being a bug on your branch causing this. Come up with a small diff (why not [#1967](https://github.com/commaai/opendbc/pull/1967)?) and try that based on latest master please.

Failure on clean 1967 on latest master was in:
ad9840558640c31d/00000080--a89435ae85

jyoung8607, 2025-07-14:
I'm given to understand you don't have this RDX loaner anymore, so there's not much we can do to move this forward. I agree with Shane that this is almost certainly not a core openpilot/opendbc bug; it's very normal for bugs in safety code to drop steering control messages and thereby trigger a fault in the EPS. If we see this again while moving the RDX port forward with other testers, obviously we can dig more into it then.

closed 2025-07-14 by jyoung8607