commaai/opendbc #2557: Enable 100% branch coverage check

adeebshihadeh, 2025-07-25:
Your goal is to [enable](https://github.com/commaai/opendbc/pull/2555) the branch coverage check and carefully fix all the violations.

For each violation:
1. can we remove this branch?
2. can we rewrite this branch?
3. add test coverage

Our test coverage should be fairly complete, so this should mostly be removing and rewriting.

Here's where we're at now:
[image]

adeebshihadeh, 2025-07-26:
> [@adeebshihadeh](https://github.com/adeebshihadeh) I would like to work on this? Can I get assigned?

You can just do things!

PhantomInTheWire, 2025-09-07:
hey @adeebshihadeh  is this issue open to work on now?

adeebshihadeh, 2025-12-17:
Locked to @mpurnell1

rekudyu, 2026-02-24:
> > Is this still available since there aren't any updates.
> 
> I'm still working on it. Here is the [latest PR](https://github.com/commaai/opendbc/pull/3148)

okay.

adeebshihadeh, 2026-09-15:
Removed bounty, it's all slop.

closed 2026-11-16 by adeebshihadeh