commaai/opendbc #2632: Honda: CAN-FD cars don't return FW or VIN over OBD

jyoung8607, 2025-08-06:
Bosch CAN-FD cars, e.g. #2610, are only getting back firmware for the camera and radar. The only successful queries are directly on the bus where those controllers live.

It appears that queries via comma power/OBD are dropped by the car's gateway before they get to the controller; we don't see the OBD queries for the camera and radar appear on that bus. It doesn't appear to be timing; it still happens with late startup. I also tried prefixing tester-present or a diagnostic session change without success.

Strangely, these queries work fine with query_fw_versions from the Panda repo. There may be some stateful filtering happening that could be getting swamped by openpilot's parallel query mechanism.

**Route:** `f39cf149898833ff/00000046--3298c279ff`

Manual query results: [query-fw-debug-output.txt](https://github.com/user-attachments/files/21634153/query-fw-debug-output.txt).

mvl-boston, 2025-08-06:
I could not find any routelogs where my Honda's returned the larger number of fingerprint ECU results.  I suspect it was early 2025 before I bought comma prime so deleted off.   I didn't use comma power but got way more results in the past.

I suggest trying your Accord on the 0.9.7 release3 diff, I recall it got more firmwares back then.   And then you can see which release dropped them to help narrow down root cause.

(I have the RDX only tonight so will try to get lockout testing done there today), but can similarly run this on my MDX tomorrow to see where all the results dropped.

jyoung8607, 2025-08-07:
- [ ] Fix root cause: commaai/panda#2256
- [ ] Track dongles used to add UDS firmware for new Honda CAN-FD cars, refresh firmware after this is fixed
- [ ] Re-check whether the EPS/VSA firmware exception is needed, think yes but make sure