commaai/opendbc #2713: Rivian fault

adeebshihadeh, 2025-08-30:
We haven't seen this in our Rivian, but apparently some users are experiencing a fault on startup.

A couple users hitting this:
* https://discord.com/channels/469524606043160576/1196513491025133638/1411490296600203326
* https://x.com/itskyleconner/status/1961926362698223857

sshane, 2025-09-03:
- Motor fault from troy: 0c832cd010dc203a/00000039--36cfb0e2a3/1
- Motor fault from rawsignal: https://connect.comma.ai/574129f33798790b/0000001f--1299806aaf

sshane, 2025-09-03:
https://www.google.com/search?q=rivian+motor+fault+detected returns many forum results of even non-comma users having similar issues. Since this can happen before the comma 3X even boots, it's unclear if the rate at which these faults occur increases with comma 3X usage.

- https://www.reddit.com/r/Rivian/comments/191jezf/motor_fault_detected/
- https://www.facebook.com/groups/RivianEV/posts/1881270055963600/
- https://www.rivianownersforum.com/threads/r1s-died-at-stop-light-mutilple-fault-codes.6223/
- https://www.rivianownersforum.com/threads/r1s-gen-2-motor-fault-and-bad-efficiency.8854/
- https://www.edmunds.com/car-news/our-long-term-2022-rivian-r1t-quits-without-warning.html
- https://www.reddit.com/r/Rivian/comments/10xixjt/motor_fault_detected/?tl=ms
- https://trailrivian.com/threads/r1s-gen1-motor-fault-and-system-fault.1037/
- https://www.torquenews.com/18000/800-miles-my-rivian-r1t-and-its-already-lost-its-rear-motor-and-gone-turtle-mode-now-im

adeebshihadeh, 2025-09-04:
Our findings so far:
* We are unable to reproduce this issue on the comma R1S. 
* Most users have not experienced this issue.
* There are plenty of instances online of non-comma users experiencing this issue.
* It's not clear what the base rate is, and whether installing a comma increases the rate at which this happens.

We'll leave this issue open while we continue to look into it, but so far this seems like a Rivian bug that a comma 3X *may* exacerbate.

---

For any users experiencing this, answering the following would be helpful:
* are you using any third party (e.g. Lukas/xnor products) hardware?
* do you have the comma power installed?
* did it happen while running stock openpilot or a fork?
* did you experience the fault before installing a comma?
* how long did you drive with your Rivian before installing the comma?

sshane, 2025-09-05:
Interesting findings from https://www.youtube.com/watch?v=QdBDgDWfrU4. Apparantly the car will wake up from deep sleep to charge the 12V battery when it gets low, also causing phantom drain in the HV battery.

> Rapid Discharge Cycles: The battery, designed for internal combustion engine vehicles, is not suitable for the demands of the Rivian R1. When the vehicle is asleep, it discharges the battery in 8 hours, then recharges it for 3 hours, repeating this cycle multiple times a day (1:25). This constant cycling wears down the battery quickly, despite it being a "deep cycle" battery (1:42).

dorkmo, 2025-09-06:
Not sure why but it looks like the battery is not reaching the peak voltage before the fault. 14.6 and 14.2 respectively. If comma comes on at the wrong moment could it be pulling down voltage slightly?

[image]

[image]